linuxhw/hw-probe #123: Excessive probing and logging

grahamperrin, 2022-03-12:
From `hw-probe -help`: 

```text
  -probe
      Probe for hardware. Collect only
      hardware related logs.
```

### Expected

With general option `-probe`, logs not related to hardware **must not** be collected.

### Actual result

Inappropriate collection of information about things such as installed software and locale. 

```text
root@mowa219-gjp4-8570p-freebsd:~ # rm -r /root/HW_PROBE/LATEST
root@mowa219-gjp4-8570p-freebsd:~ # hw-probe -probe && cd /root/HW_PROBE/LATEST && du -hs hw.info && ls -R hw.info && cd
Probe for hardware ... Ok
Reading logs ... Ok
Local probe path: /root/HW_PROBE/LATEST/hw.info
581K    hw.info
devices.json    host            logs

hw.info/logs:
apm             dmesg           gpart           locale          osname          sysctl
biosdecode      dmidecode       gpart_list      lscpu           pciconf         uname
dev             freebsd-version hwstat          lspci           pkglist         usbconfig
devinfo         geom            ifconfig        lsusb           smartctl        xorg.log
df              glxinfo         kldstat         mcelog          sndstat         xrandr
root@mowa219-gjp4-8570p-freebsd:~ # head -n5 /root/HW_PROBE/LATEST/hw.info/logs/pkglist          FreeBSD accessibility/accerciser 3.38.0
FreeBSD accessibility/at-spi2-atk 2.34.2
FreeBSD accessibility/at-spi2-core 2.36.0
FreeBSD accessibility/atk 2.36.0
FreeBSD accessibility/atkmm 2.28.0
root@mowa219-gjp4-8570p-freebsd:~ # cat /root/HW_PROBE/LATEST/hw.info/logs/locale                LANG=C.UTF-8
LC_CTYPE="C"
LC_COLLATE="C"
LC_TIME="C"
LC_NUMERIC="C"
LC_MONETARY="C"
LC_MESSAGES="C"
LC_ALL=C
root@mowa219-gjp4-8570p-freebsd:~ # rm -r /root/HW_PROBE/LATEST
root@mowa219-gjp4-8570p-freebsd:~ # hw-probe -dumpversion
1.6
root@mowa219-gjp4-8570p-freebsd:~ # pkg info -x hw-probe
hw-probe-1.6.1
root@mowa219-gjp4-8570p-freebsd:~ # uname -aKU
FreeBSD mowa219-gjp4-8570p-freebsd 14.0-CURRENT FreeBSD 14.0-CURRENT #5 main-n253627-25375b1415f-dirty: Sat Mar  5 14:21:40 GMT 2022     root@mowa219-gjp4-8570p-freebsd:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64 1400053 1400053
root@mowa219-gjp4-8570p-freebsd:~ # 
```

linuxhw, 2022-03-15:
I'll remove this obsolete option in the next commit.

Thank you!

incrediblyimpressiveusername, 2024-04-19:
Despite the privacy statement saying "Private information (including the username, machine's hostname, IP addresses, MAC addresses, UUIDs and serial numbers) is NOT uploaded to the database.", the recommended probe (`sudo -E hw-probe -all -upload`) still collects and uploads excessive private and personally identifiable information, including but not limited to the partition table, network configuration and a full list of installed packages!

This was indicated nowhere on https://linux-hardware.org/?view=howto and the contrary was implied. I sent you an E-mail requesting deletion of my probe under the GDPR right to removal, but you are block E-mails from Proton Mail, so I can't even exercise my GDPR rights without being forced to create a GitHub account.

Please unblock Proton Mail so I can privately tell you which probe to delete.
And please update https://linux-hardware.org/?view=howto to reflect that `sudo -E hw-probe -all -upload` most certainly does upload private information to the database.

@linuxhw

grahamperrin, 2024-11-29:
> I'll remove this obsolete option in the next commit.
> 
> Thank you!

Without me looking at commit history, I can see that the option is no longer present in 1.6.5. 

Thanks

```text
% pkg iinfo hw-probe
hw-probe-1.6.5
% hw-probe -help | grep hardware\ related
% 
```

grahamperrin, 2025-04-20:
Found:

> Remove obsolete options -probe and -logs · linuxhw/hw-probe@8c3a517

– 2022-03-21