linuxhw/hw-probe #124: log level ambiguities

grahamperrin, 2022-03-12:
### Observation A

From `hw-probe -help`: 

```text
  -minimal|-min
      Collect minimal number of logs. Equal to --log-level=min.
  
  -maximal|-max
      Collect maximal number of logs. Equal to --log-level=max.
```

### Issue

There is no `--log-level` option in help. 

### Observation B

```text
  -logs
      Collect system logs.
  
  -log-level N
      Set the logging level to N.
      Available values:
      
        - minimal
        - default
        - maximal
```

### Issue

In the observation A, there's mention of `=` for use with `--log-level` (two hyphens).

In the observation B, there's space (not `=`) between `-log-level` (one hyphen) and the value. 

### Context

```text
root@mowa219-gjp4-8570p-freebsd:~ # uname
FreeBSD
root@mowa219-gjp4-8570p-freebsd:~ # pkg info -x hw-probe
hw-probe-1.6.1
root@mowa219-gjp4-8570p-freebsd:~ # hw-probe -help
```

<details>
<summary>hw-probe -help</summary>
    
    NAME:
    Hardware Probe (hw-probe)
    A tool to probe for hardware, check operability and find drivers

    DESCRIPTION:
    Hardware Probe (hw-probe) is a tool to probe for hardware,
    check its operability and upload result to the BSD hardware database.
    
    By creating probes you contribute to the "HDD/SSD Desktop-Class Reliability
    Test" study: https://github.com/linuxhw/SMART

    USAGE:
    hw-probe [options]

    EXAMPLE:
    hw-probe -all -upload

    PRIVACY:
    Private information (including the username, machine's hostname, IP addresses,
    MAC addresses, UUIDs and serial numbers) is NOT uploaded to the database.
    
    The tool uploads 32-byte prefix of salted SHA512 hash of MAC addresses and serial
    numbers to properly identify unique computers and hard drives. UUIDs are decorated
    in the same way, but formatted like regular UUIDs in order to save readability of
    logs. All the data is uploaded securely via HTTPS.

    INFORMATION OPTIONS:
    -h|-help
        Print this help.
    
    -v|-version
        Print version info.

    -dumpversion
        Print the tool version (1.6) and don't do anything else.

    GENERAL OPTIONS:
    -all
        Enable all probes.
    
    -probe
        Probe for hardware. Collect only
        hardware related logs.
    
    -logs
        Collect system logs.
    
    -log-level N
        Set the logging level to N.
        Available values:
        
            - minimal
            - default
            - maximal
    
    -minimal|-min
        Collect minimal number of logs. Equal to --log-level=min.
    
    -maximal|-max
        Collect maximal number of logs. Equal to --log-level=max.
    
    -enable LIST
        Comma separated list of logs to enable in addition to
        current log level.
    
    -disable LIST
        Comma separated list of logs to disable in current
        log level. Some logs cannot be disabled. For example,
        you can disable collecting of 'fstab', but you cannot
        disable logging of 'smartctl'.
    
    -printers
        Probe for printers.
    
    -scanners
        Probe for scanners.
    
    -check
        Check devices operability.
    
    -id|-name DESC
        Any description of the probe.
    
    -upload
        Upload result to the hardware database. You will get
        a permanent URL to view the probe.
        
        By use of this option you confirm uploading of 32-byte
        prefix of salted SHA512 hash of MAC addresses and serial
        numbers to prevent duplication of computers in the DB.
    
    -hwinfo-path PATH
        Path to a local hwinfo binary.
    
    -proxy ADDRESS:PORT
        Set outgoing http/https proxy using syntax: proxy.domain.local:3128

    INVENTORY OPTIONS:
    -i|-inventory-id ID
        Mark the probe by inventory ID.
    
    -generate-inventory-id
        Generate new inventory ID.
    
    -email ADDR
        Email for node status notifications.

    MONITORING OPTIONS:
    -start
        Start monitoring of the node.
    
    -stop
        Stop monitoring of the node.
    
    -remind-inventory
        Remind node inventory ID.

    OTHER OPTIONS:
    -save DIR
        Save probe package to DIR. This is useful if you are offline
        and need to upload a probe later (with the help of -src option).
    
    -src|-source PATH
        A probe to upload.
    
    -fix PATH
        Update list of devices and host info
        in the probe using probe data.
    
    -show-devices
        Show devices list.
    
    -show
        Show host info and devices list.
    
    -show-host
        Show host info only.
    
    -verbose
        Use with -show option to show type and status of the device.
    
    -pci-ids  PATH
    -usb-ids  PATH
    -sdio-ids PATH
    -pnp-ids  PATH
        Path to {pci,usb,sdio,pnp}.ids file to read missed device names.
    
    -list
        List executed probes (for debugging).
    
    -clean
        Do nothing. Obsolete option.
    
    -save-uploaded
        Save uploaded probes.
    
    -debug|-d
        Do nothing. Obsolete option.
    
    -dump-acpi
        Probe for ACPI table.
    
    -decode-acpi
        Decode ACPI table.
    
    -import DIR
        Import probes from the database to DIR for offline use.
        
        If you are using Snap or Flatpak package, then DIR will be created
        in the sandbox data directory.
        
        Provide inventory ID by -i option in order to import your inventory.

    DATA LOCATION:
    Probes info is saved in the /root/HW_PROBE directory.

</details>

grahamperrin, 2022-03-12:
See also: #123 

Disk usage from six runs: 

1. `hw-probe -logs -log-level minimal` 621K
2. `hw-probe -logs -log-level=minimal` _unknown option_
3. `hw-probe -logs --log-level minimal` 633K
4. `hw-probe -logs -log-level maximal` 660K
5. `hw-probe -logs --log-level maximal` 660K
6. `hw-probe -logs --log-level max` 732K

In detail: 

```text
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # rm -r *
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # hw-probe -logs -log-level minimal && du -hs hw.info && ls -R hw.info && rm -r hw.info
Probe for hardware ... Ok
Reading logs ... Ok
Local probe path: /root/HW_PROBE/LATEST/hw.info
621K    hw.info
devices.json    host            logs

hw.info/logs:
apm             dmesg           gpart           locale          osname          sysctl
biosdecode      dmidecode       gpart_list      lscpu           pciconf         uname
dev             freebsd-version hwstat          lspci           pkglist         usbconfig
devinfo         geom            ifconfig        lsusb           smartctl        xorg.log
df              glxinfo         kldstat         mcelog          sndstat         xrandr
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # hw-probe -logs -log-level=minimal && du -hs hw.info && ls -R hw.info && rm -r hw.info
Unknown option: log-level=minimal

Hardware Probe 1.6
A tool to probe for hardware, check operability and find drivers
License: LGPL-2.1-or-later OR BSD-4-Clause

Usage: hw-probe [options]
Example: hw-probe -all -upload

root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # hw-probe -logs --log-level minimal && du -hs hw.info && ls -R hw.info && rm -r hw.info
Probe for hardware ... Ok
Reading logs ... Ok
Local probe path: /root/HW_PROBE/LATEST/hw.info
633K    hw.info
devices.json    host            logs

hw.info/logs:
apm             dmesg           gpart           locale          osname          sysctl
biosdecode      dmidecode       gpart_list      lscpu           pciconf         uname
dev             freebsd-version hwstat          lspci           pkglist         usbconfig
devinfo         geom            ifconfig        lsusb           smartctl        xorg.log
df              glxinfo         kldstat         mcelog          sndstat         xrandr
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # hw-probe -logs -log-level maximal && du -hs hw.info && ls -R hw.info && rm -r hw.info
Probe for hardware ... Ok
Reading logs ... Ok
Local probe path: /root/HW_PROBE/LATEST/hw.info
660K    hw.info
devices.json    host            logs

hw.info/logs:
apm                     geom                    lspci                   top_head
biosdecode              glxinfo                 lspci_all               uname
camcontrol              gpart                   lsusb                   uptime
config                  gpart_list              mcelog                  usbconfig
dev                     hwstat                  mount                   vmstat
devinfo                 ifconfig                neofetch                x86info
df                      iostat                  osname                  xinput
diskinfo                kldstat                 pciconf                 xorg.conf
dmesg                   kldstat_v               pkglist                 xorg.conf.d
dmidecode               loader.conf             rc.conf                 xorg.log
drm_info                locale                  smartctl                xorg.log.1
[... 21 lines cut ...]
freebsd-version         lsblk                   sndstat                 xrandr
fstab                   lscpu                   sysctl                  xrandr_providers
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # hw-probe -logs --log-level max && du -hs hw.info && ls -R hw.info && rm -r hw.info
Probe for hardware ... Ok
Reading logs ... Ok
Local probe path: /root/HW_PROBE/LATEST/hw.info
732K    hw.info
devices.json    host            logs

hw.info/logs:
apm                     geom                    lspci                   top_head
biosdecode              glxinfo                 lspci_all               uname
camcontrol              gpart                   lsusb                   uptime
config                  gpart_list              mcelog                  usbconfig
dev                     hwstat                  mount                   vmstat
devinfo                 ifconfig                neofetch                x86info
df                      iostat                  osname                  xinput
diskinfo                kldstat                 pciconf                 xorg.conf
dmesg                   kldstat_v               pkglist                 xorg.conf.d
dmidecode               loader.conf             rc.conf                 xorg.log
drm_info                locale                  smartctl                xorg.log.1
freebsd-version         lsblk                   sndstat                 xrandr
fstab                   lscpu                   sysctl                  xrandr_providers
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # 
```

linuxhw, 2022-03-15:
The data in this directory is cleaned only after `-upload` or `-save` actions. So each new run w/o `-upload` and `-save` adds more logs in this directory.

Argument

    --option=VAL
    
is equivalent of:

    -option VAL

And this is always invalid:

    -option=VAL

This is standard for all Linux and BSD systems since the beginning of time.

> The Getopt::Long module implements an extended getopt function called GetOptions(). It parses the command line from @ARGV, recognizing and removing specified options and their possible values.
>
>This function adheres to the POSIX syntax for command line options, with GNU extensions. In general, this means that options have long names instead of single letters, and are introduced with a double dash "--".

See https://perldoc.perl.org/Getopt::Long

grahamperrin, 2022-03-16:
Thanks for looking at these issues. 

Would it help to expand the lines just a little, for example:

          Collect minimal number of logs. Equal to --log-level=min in Perl.

– then people (like me) who have no interest in (or understanding of) Perl can know, at a glance, to either: 

a) ignore the sentence about Perl; or

b) learn more, if there's interest. 

(Does that make sense?)

linuxhw, 2022-04-01:
It's not Perl-specific syntax. It's Unix-specific syntax for all programs implemented in any language.