linuxhw/hw-probe #126: Extraordinarily large /root/HW_PROBE/LATEST/hw.info/logs/sysctl preventing upload

grahamperrin, 2022-03-12:
<https://forums.freebsd.org/posts/558998> a few days ago: 

----

[sysutils/hw-probe](http://www.freshports.org/sysutils/hw-probe) failed upload to upload probe …

In recent weeks, I'm unable to upload the result of any probe from my usual computer.

Has anyone else encountered the symptom below?

```text
ERROR: the probe is larger than 500K

ERROR: failed to upload probe
```

The offending file:

```text
% ls -hl /root/HW_PROBE/LATEST/hw.info/logs/sysctl
-rw-r--r--  1 root  wheel   2.0M  6 Mar 12:35 /root/HW_PROBE/LATEST/hw.info/logs/sysctl
% du -hs /root/HW_PROBE/LATEST/hw.info/logs/sysctl
521K    /root/HW_PROBE/LATEST/hw.info/logs/sysctl
```

----

Today, for example: 

```text
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # ls -hl
total 0
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # hw-probe -logs -log-level minimal
Probe for hardware ... Ok
Reading logs ... Ok
Local probe path: /root/HW_PROBE/LATEST/hw.info
root@mowa219-gjp4-8570p-freebsd:~/HW_PROBE/LATEST # cd
root@mowa219-gjp4-8570p-freebsd:~ # tail -n 100 /root/HW_PROBE/LATEST/hw.info/logs/sysctl
24892 2739 62001007056340 62001012206575 "/sbin/sysctl" "/sbin/sysctl"
24893 2739 62003909040367 62003912396116 "/sbin/sysctl" "/sbin/sysctl"
24894 2739 62006818431516 62006821934619 "/sbin/sysctl" "/sbin/sysctl"
24895 2739 62009837406843 62009840864901 "/sbin/sysctl" "/sbin/sysctl"
24896 2739 62012750214992 62012753549168 "/sbin/sysctl" "/sbin/sysctl"
24897 2739 62015655948839 62015659294492 "/sbin/sysctl" "/sbin/sysctl"
24898 2739 62018583694027 62018587174738 "/sbin/sysctl" "/sbin/sysctl"
24899 2739 62021517563261 62021520874039 "/sbin/sysctl" "/sbin/sysctl"
24900 2739 62024419890212 62024423194109 "/sbin/sysctl" "/sbin/sysctl"
24901 2739 62027337421497 62027340776223 "/sbin/sysctl" "/sbin/sysctl"
24902 2739 62030253089919 62030257139769 "/sbin/sysctl" "/sbin/sysctl"
24903 2739 62033195571384 62033198901680 "/sbin/sysctl" "/sbin/sysctl"
24904 2739 62036137771311 62036141137802 "/sbin/sysctl" "/sbin/sysctl"
24905 2739 62039057094948 62039062574071 "/sbin/sysctl" "/sbin/sysctl"
24906 2739 62041967702283 62041971096572 "/sbin/sysctl" "/sbin/sysctl"
24907 2739 62044870851864 62044874201499 "/sbin/sysctl" "/sbin/sysctl"
24908 2739 62047774413885 62047777896840 "/sbin/sysctl" "/sbin/sysctl"
24909 2739 62050700488544 62050703858449 "/sbin/sysctl" "/sbin/sysctl"
24910 2739 62053601752469 62053605051549 "/sbin/sysctl" "/sbin/sysctl"
24911 2739 62056536722966 62056543025963 "/sbin/sysctl" "/sbin/sysctl"
24912 2739 62059473419441 62059477780363 "/sbin/sysctl" "/sbin/sysctl"
24913 2739 62062483677487 62062488751138 "/sbin/sysctl" "/sbin/sysctl"
24914 2739 62065551729071 62065556810113 "/sbin/sysctl" "/sbin/sysctl"
24915 2739 62068493812772 62068499366266 "/sbin/sysctl" "/sbin/sysctl"
24916 2739 62071470926416 62071477129248 "/sbin/sysctl" "/sbin/sysctl"
24917 2739 62074378678007 62074385683985 "/sbin/sysctl" "/sbin/sysctl"
24918 2739 62077302763914 62077308986828 "/sbin/sysctl" "/sbin/sysctl"
24919 2739 62080212412067 62080219406274 "/sbin/sysctl" "/sbin/sysctl"
24920 2739 62083212374173 62083219350917 "/sbin/sysctl" "/sbin/sysctl"
24921 2739 62086119532694 62086126532664 "/sbin/sysctl" "/sbin/sysctl"
24922 2739 62089060229119 62089067233995 "/sbin/sysctl" "/sbin/sysctl"
24923 2739 62091979388912 62091982699409 "/sbin/sysctl" "/sbin/sysctl"
[... 66 lines cut ...]
24990 2739 62287466671352 62287470181743 "/sbin/sysctl" "/sbin/sysctl"
24991 2739 62290369040072 62290372408889 "/sbin/sysctl" "...root@mowa219-gjp4-8570p-freebsd:~ # 
root@mowa219-gjp4-8570p-freebsd:~ # 
```

Maybe not an issue with hw-probe itself, but (sorry) I struggle to understand the meanings of log entries such as those. 

### Environment

```text
root@mowa219-gjp4-8570p-freebsd:~ # date ; uptime ; uname -aKU
Sat Mar 12 07:27:14 GMT 2022
 7:27AM  up 17:12, 5 users, load averages: 0.45, 0.77, 0.83
FreeBSD mowa219-gjp4-8570p-freebsd 14.0-CURRENT FreeBSD 14.0-CURRENT #5 main-n253627-25375b1415f-dirty: Sat Mar  5 14:21:40 GMT 2022     root@mowa219-gjp4-8570p-freebsd:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64 1400053 1400053
root@mowa219-gjp4-8570p-freebsd:~ # pkg info -x hw-probe
hw-probe-1.6.1
root@mowa219-gjp4-8570p-freebsd:~ # pkg -vv | grep -e url -e enabled
    url             : "http://pkg0.bme.freebsd.org/FreeBSD:14:amd64/latest",
    enabled         : yes,
    url             : "https://alpha.pkgbase.live/current/FreeBSD:14:amd64/latest",
    enabled         : no,
    url             : "file:///usr/local/poudriere/data/packages/main-default",
    enabled         : yes,
root@mowa219-gjp4-8570p-freebsd:~ # 
```

grahamperrin, 2022-03-12:
Are these lines (from the most recent commit) relevant? 

https://github.com/linuxhw/hw-probe/blob/fc68788468351138477f6ad71d6f4cf717e3810b/hw-probe.pl#L5258-L5285

If not please hide as off-topic.

linuxhw, 2022-03-15:
Could you please attach complete compressed output of `sysctl -a` here?

grahamperrin, 2022-03-20:
I _can_ attach, if you like, although I might have identified the underlying cause. 

From <https://github.com/freebsd/freebsd-src/commit/46dd801acb2318c8464b185893bb111f09b71138>

> … Expose this information via a new sysctl, debug.tslog_user. …

Here: 

```text
% date 
Sun 20 Mar 2022 14:32:28 GMT
% tail -n 1 /usr/src/sys/amd64/conf/GENERIC-NODEBUG
options         TSLOG
% tail -n 1 /usr/src/sys/amd64/conf/GENERIC
options         TSLOG
% 
```

<https://old.reddit.com/r/freebsd/comments/rya5pz/-/hsy73fi/?context=1> and <https://old.reddit.com/r/freebsd/comments/s67ymv/-/> remind me that my first complete build of a TSLOG-enabled kernel was on 2022-01-17. 

The preceding probe (2022-01-15) has a record that is relatively short: 

* <https://bsd-hardware.info/?probe=5f106ee686&log=sysctl&d=freebsd>

– the three subsequent probes have records that are relatively long: 

* <https://bsd-hardware.info/?probe=1520fece28&log=sysctl&d=freebsd>
* <https://bsd-hardware.info/?probe=61406080a7&log=sysctl&d=freebsd>
* <https://bsd-hardware.info/?probe=f47789d894&log=sysctl&d=freebsd>

– and truncated, ending with `...`

Now, this makes sense: 

[image]

grahamperrin, 2022-03-20:
@cperciva just FYI, above <https://github.com/linuxhw/hw-probe/issues/126#issuecomment-1073269062>

@linuxhw I will avoid uploading such large files: 

`hw-probe -all -disable sysctl -upload`

– result (no sysctl): <https://bsd-hardware.info/?probe=7e1e137c8f>

cperciva, 2022-03-20:
@grahamperrin Right, TSLOG exports a bunch of data via sysctls.  But now that you mention it, I think I can set a flag on those so they don't show up with `sysctl -a`.

cperciva, 2022-03-20:
Fixed in FreeBSD HEAD.

linuxhw, 2022-04-01:
Fixed by 8c3a5174208ffa348cb38337537408ed2996ef91. Part of 1.6.3 release.

The debug.tslog_user entry in the sysctl is now removed before uploading.

Thanks!

closed 2022-04-01 by linuxhw

grahamperrin, 2022-04-02:
Excellent! 

Thanks, people. 

https://github.com/freebsd/freebsd-src/commit/2406867f5b178a81c653b14524c0d5bad8189d2c