linuxhw/hw-probe #147: API / curl

ldrahnik, 2023-01-19:
I would like to connect the install script of the [developed driver](https://github.com/asus-linux-drivers/asus-touchpad-numpad-driver/issues/63) to the database. What is the best approach? (bash) Thank you.

ldrahnik, 2023-01-21:
Used `.. --user-agent "user-agent-name-here" ..` as workaround.

closed 2023-01-21 by ldrahnik

linuxhw, 2023-03-11:
Hello

Info for ASUE1403 can be quickly found with the help of auxiliary repo https://github.com/linuxhw/Dmesg

Clone it and grep what you want. Result for ASUE1403 is the following:

```
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:942:[    1.404147] input: ASUE1403:00 04F3:319A Mouse as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input9
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:943:[    1.404228] input: ASUE1403:00 04F3:319A Touchpad as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input10
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:944:[    1.404289] input: ASUE1403:00 04F3:319A Keyboard as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input11
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:945:[    1.404408] hid-generic 0018:04F3:319A.0002: input,hidraw1: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:954:[    1.595630] input: ASUE1403:00 04F3:319A Mouse as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input12
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:955:[    1.694347] input: ASUE1403:00 04F3:319A Touchpad as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input13
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:956:[    1.711169] input: ASUE1403:00 04F3:319A Keyboard as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0002/input/input14
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/A28CBA5E0529/FEDORA-36/5.18.5-200.FC36.X86_64/X86_64/3B78B34416:957:[    1.728191] hid-multitouch 0018:04F3:319A.0002: input,hidraw0: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:957:[    0.880093] input: ASUE1403:00 04F3:319A Mouse as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input5
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:958:[    0.880306] input: ASUE1403:00 04F3:319A Touchpad as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input6
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:959:[    0.880413] input: ASUE1403:00 04F3:319A Keyboard as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input7
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:960:[    0.880491] hid-generic 0018:04F3:319A.0001: input,hidraw0: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:1227:[    4.333701] input: ASUE1403:00 04F3:319A Mouse as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input29
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:1229:[    4.333768] input: ASUE1403:00 04F3:319A Touchpad as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input30
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:1230:[    4.333876] input: ASUE1403:00 04F3:319A Keyboard as /devices/platform/AMDI0010:00/i2c-0/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input31
./ASUSTek Computer/ROG/ROG Strix G513QM_G513QM/9CC7E5240E82/LINUXMINT-20.3/5.13.0-39-GENERIC/X86_64/BDCA066BA3:1231:[    4.334129] hid-multitouch 0018:04F3:319A.0001: input,hidraw0: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
./ASUSTek Computer/ROG/ROG Strix G533ZM_G533ZM/F648E3BD7DA2/POP!_OS-21.10/5.16.15-76051615-GENERIC/X86_64/4A159B7CD8:1029:[    1.858668] input: ASUE1403:00 04F3:319A Mouse as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input4
./ASUSTek Computer/ROG/ROG Strix G533ZM_G533ZM/F648E3BD7DA2/POP!_OS-21.10/5.16.15-76051615-GENERIC/X86_64/4A159B7CD8:1030:[    1.858841] input: ASUE1403:00 04F3:319A Touchpad as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input5
./ASUSTek Computer/ROG/ROG Strix G533ZM_G533ZM/F648E3BD7DA2/POP!_OS-21.10/5.16.15-76051615-GENERIC/X86_64/4A159B7CD8:1031:[    1.858984] input: ASUE1403:00 04F3:319A Keyboard as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input6
./ASUSTek Computer/ROG/ROG Strix G533ZM_G533ZM/F648E3BD7DA2/POP!_OS-21.10/5.16.15-76051615-GENERIC/X86_64/4A159B7CD8:1032:[    1.859052] hid-generic 0018:04F3:319A.0001: input,hidraw0: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
[... cut ...]
./ASUSTek Computer/ROG/ROG Strix G533ZW_G533ZW/560A787C4010/UBUNTU-22.04/5.19.0-051900-GENERIC/X86_64/FFAB2B901D:1445:[    5.509800] input: ASUE1403:00 04F3:319A Keyboard as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-ASUE1403:00/0018:04F3:319A.0001/input/input32
./ASUSTek Computer/ROG/ROG Strix G533ZW_G533ZW/560A787C4010/UBUNTU-22.04/5.19.0-051900-GENERIC/X86_64/FFAB2B901D:1448:[    5.510173] hid-multitouch 0018:04F3:319A.0001: input,hidraw7: I2C HID v1.00 Mouse [ASUE1403:00 04F3:319A] on i2c-ASUE1403:00
```