linuxhw/hw-probe #148: Docker container linuxhw/hw-probe to unreachable host https://linux-hardware.ru/

sarvasana, 2023-01-21:
The [docker example](https://github.com/linuxhw/hw-probe#docker) is uploading to unreachable host https://linux-hardware.ru/.

Looks wrong. I don't trust any .ru domain.

[image]

sudo -E docker run -it \
-v /dev:/dev:ro \
-v /lib/modules:/lib/modules:ro \
-v /etc/os-release:/etc/os-release:ro \
-v /var/log:/var/log:ro \
--privileged --net=host --pid=host \
linuxhw/hw-probe -all -upload

linuxhw, 2023-04-11:
This is because the global endpoint linux-hardware.org is not accessible from your location for some reason. In your case, the reserved endpoint in Eastern Europe was selected temporarily.

closed 2023-04-11 by linuxhw

sarvasana, 2023-04-11:
Well, it worked when installed in my machine directly instead of in docker. Must have been docker networks not reaching something. Still puzzled why you would host anything in Russia at all.