linuxhw/hw-probe #161: Feature: Provide option to remove uploaded submission

michaelkrieger, 2023-08-04:
As per the subject, sometimes we -upload to the Linux Hardware Database and see something we don't like.

Requesting the ability to, from the machine which submitted the original upload, remove a submission.

Kev-Griffin, 2023-09-08:
I second that.

Can you please also remove my probe [https://linux-hardware.org/?probe=7dd7a13a5e](https://linux-hardware.org/?probe=7dd7a13a5e)?

incrediblyimpressiveusername, 2024-04-19:
Based on [your privacy policy](https://linux-hardware.org/?view=howto) (https://linux-hardware.org/?view=howto) stating that "_private information is NOT uploaded to the database_", I had not expected that such private and personally identifiable information as my network configuration, partition table, installed packages, etc. would be uploaded to the database.

Under the present circumstances, being able to scrub such probe data from the database is vital.

I would like to exercise my right to removal under the GDPR, to request the deletion of my probe. However, when I sent you an E-mail to that effect, I received "Undelivered Mail Returned to Sender. 550 Message rejected. Protonmail is blocked"

Please unblock Proton Mail so I can privately tell you which probe to delete, without giving out the data from the probe to everyone in the meanwhile.
Please then delete all copies of that probe, both from the public database and any other private copies that may be stored elsewhere.
And please update https://linux-hardware.org/?view=howto to reflect which private information `sudo -E hw-probe -all -upload` DOES upload to the database.

@lvc @linuxhw 

See also https://github.com/linuxhw/hw-probe/issues/123#issuecomment-2065958184

grahamperrin, 2024-06-02:
> … remove my probe …?

404

RokeJulianLockhart, 2024-11-29:
https://github.com/linuxhw/hw-probe/issues/145#issue-1501905125 is relevant.