linuxhw/hw-probe #187: Inventory email failing on SMTP Auth

JVilleComputers, 2025-01-28:
root@MXDEV:~# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

root@MXDEV:~# dpkg -l hw-probe
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-==============================================
ii  hw-probe       1.6.5-1      all          Tool to check operability of computer hardware

root@MXDEV:~#  hw-probe -generate-inventory -email [REDACT1]
Inventory ID: [REDACT2]

root@MXDEV:~# hw-probe -all -upload -i [REDACT2]
Probe for hardware ... Ok
Reading logs ... Ok
failed AUTH: 5.7.0 NEOBHODIM parol prilozheniya https://help.mail.ru/mail/security/protection/external / Application password is REQUIRED

Trace begun at /usr/share/perl5/vendor_perl/Email/Sender/Transport/SMTP.pm line 232
Email::Sender::Transport::SMTP::_throw('Email::Sender::Transport::SMTP=HASH(0x562b6619c728)', 'failed AUTH', 'Net::SMTP::_SSL=GLOB(0x562b6720fb48)') called at /usr/share/perl5/vendor_perl/Email/Sender/Transport/SMTP.pm line 204
Email::Sender::Transport::SMTP::_smtp_client('Email::Sender::Transport::SMTP=HASH(0x562b6619c728)') called at /usr/share/perl5/vendor_perl/Email/Sender/Transport/SMTP.pm line 241
Email::Sender::Transport::SMTP::send_email('Email::Sender::Transport::SMTP=HASH(0x562b6619c728)', 'Email::Abstract=ARRAY(0x562b66dff068)', 'HASH(0x562b670508b8)') called at /usr/share/perl5/vendor_perl/Email/Sender/Role/CommonSending.pm line 45
Email::Sender::Role::CommonSending::try {...}  at /usr/share/perl5/vendor_perl/Try/Tiny.pm line 100
eval {...} at /usr/share/perl5/vendor_perl/Try/Tiny.pm line 93
Try::Tiny::try('CODE(0x562b671fb7e0)', 'Try::Tiny::Catch=REF(0x562b671b8cd0)') called at /usr/share/perl5/vendor_perl/Email/Sender/Role/CommonSending.pm line 58
Email::Sender::Role::CommonSending::send('Email::Sender::Transport::SMTP=HASH(0x562b6619c728)', 'Email::Abstract=ARRAY(0x562b66dff068)', 'HASH(0x562b671fb138)') called at /usr/share/perl5/vendor_perl/Email/Sender/Simple.pm line 119
Email::Sender::Simple::send_email('Email::Sender::Simple', 'Email::Abstract=ARRAY(0x562b66dff068)', 'HASH(0x562b671b7bf0)') called at /usr/share/perl5/vendor_perl/Email/Sender/Role/CommonSending.pm line 45
Email::Sender::Role::CommonSending::try {...}  at /usr/share/perl5/vendor_perl/Try/Tiny.pm line 100
eval {...} at /usr/share/perl5/vendor_perl/Try/Tiny.pm line 93
Try::Tiny::try('CODE(0x562b67190478)', 'Try::Tiny::Catch=REF(0x562b66b8e940)') called at /usr/share/perl5/vendor_perl/Email/Sender/Role/CommonSending.pm line 58
Email::Sender::Role::CommonSending::send('Email::Sender::Simple', 'Email::Simple=HASH(0x562b67190688)', 'HASH(0x562b670508a0)') called at device-checker.pl line 3685
main::sendLetter(1, 'HASH(0x562b660fe748)', 'smtp.mail.ru', 465, 'info@linux-hardware.org', 'LSg17VYNR4eRCKCuteDL') called at device-checker.pl line 3650
main::sendLetters('[REDACT2]') called at device-checker.pl line 3516
main::addComputerToGroup('[REDACT2]', 'af76189b4b4156c44e0beb868745785e', '[REDACT3]', 'HASH(0x562b660fee50)') called at device-checker.pl line 3488
main::addProbe('/var/lib/LHW/PHP_TEMP/php8M4Hqz') called at device-checker.pl line 19518
main::scenario at device-checker.pl line 19742
Uploaded to DB, Thank you!

Probe URL: https://linux-hardware.org/?probe=[REDACT3]

JVilleComputers, 2025-01-28:
This leads to the additional issue of new computers now showing up in Inventory List, due to not being able to confirm from email.

[image]

taosecurity, 2025-05-20:
I had the same issue just now on Linux. It WORKED on FreeBSD, however. I got a confirmation email from <info@bsd-hardware.info> for this FreeBSD system.

https://bsd-hardware.info/?probe=6687e92476

wblondel, 2025-05-20:
I confirm it's still broken. I can't add Debian probes.

taosecurity, 2025-05-21:
Copy that. Mine were Raspberry OS and Ubuntu.